RosLibRust/roslibrust #27: Error when trying to implement turtlesim node

dariandzirko, 2022-08-31:
On ubuntu 20.4 with ROS noetic install.

Tried making the following changes to basic_publisher.rs example
```rs
+ use geometry_msgs::{Twist, Vector3};
- roslibrust_codegen_macro::find_and_generate_ros_messages!("roslibrust/local_msgs");
+ roslibrust_codegen_macro::find_and_generate_ros_messages!();

-     let publisher = client.advertise::<std_msgs::Header>("talker").await?;
+    let publisher = client.advertise::<Twist>("/turtle1/cmd_vel").await?;
    loop {
-         let msg = std_msgs::Header::default();
+        let msg = Twist {
+            linear: Vector3 {
+                x: 0.0,
+                y: 0.0,
+                z: 0.0,
+            },
+            angular: Vector3 {
+               x: 1.0,
+                y: 0.0,
+                z: 0.0,
+            },
+        };
```
This was the error output from the console when running the following
``console
cargo run --example basic_publisher
```

**console output**

```console
error[E0412]: cannot find type `DurationI` in module `std_msgs`
 --> roslibrust/examples/basic_publisher.rs:4:1
  |
4 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `std_msgs`
  |
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0412]: cannot find type `TimeI` in module `std_msgs`
 --> roslibrust/examples/basic_publisher.rs:4:1
  |
4 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `std_msgs`
  |
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0412]: cannot find type `DurationI` in module `self`
 --> roslibrust/examples/basic_publisher.rs:4:1
  |
4 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `self`
  |
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0412]: cannot find type `TimeI` in module `self`
 --> roslibrust/examples/basic_publisher.rs:4:1
  |
4 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `self`
  |
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0283]: type annotations needed
 --> roslibrust/examples/basic_publisher.rs:4:1
  |
4 | roslibrust_codegen_macro::find_and_generate_ros_messages!();
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ cannot infer type
  |
  = note: cannot satisfy `_: Default`
  = note: this error originates in the derive macro `Default` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0283, E0412.
For more information about an error, try `rustc --explain E0283`.
```
The text file would have been a .rs if let me paste that in here, but output.txt is the printed output of source from roslibrust_test/main.rs

[output.txt](https://github.com/Carter12s/roslibrust/files/9464700/output.txt)

Carter12s, 2022-09-01:
Nice! Thanks for flag.

Yup this is caused by the recent re-write of our codegen which is now not always generating definitions for those types.

Fix should be:
- Create a new module in main crate `integral_types` and create definitions for Time and Duration in there
- Remove the local_msgs/std_msgs/TimeI and DurationI files
- Update codegen to generate `::roslibrust::integral_types::Time` types when it encounters the Time and Duration primitives

I'll plan to tackle this tomorrow.

Carter12s, 2022-09-01:
@dariandzirko I've filed PR: https://github.com/Carter12s/roslibrust/pull/30 which I believe will address the issue.

If you have a moment to try it out and confirm that it resolves the problem I would appreciate it.

closed 2022-09-01 by Carter12s

Carter12s, 2022-09-01:
Fix for this was released in v0.2.0

PR #30 (2022-09-01): Integral types for time duration
Closes #27