RosLibRust/roslibrust #54: conflicting implementation for `realsense2_camera::DeviceInfo`

Akumar201, 2022-11-03:
Hi, I am trying to write a publisher to publish data to another computer where ROS will be running. However, I am getting the following error. I am sharing the code with you. 

```
use log::*;
use roslibrust::ClientHandle;
use roslibrust_codegen_macro::find_and_generate_ros_messages;

roslibrust_codegen_macro::find_and_generate_ros_messages!();

/// This example creates a client, and publishes a message to the topic "talker"
/// Running this example at the same time as subscribe_and_log will have the two examples
/// pass messages between each other.
/// To run this example a rosbridge websocket server should be running at the deafult port (9090).
#[tokio::main(flavor = "multi_thread")]
async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
    simple_logger::SimpleLogger::new()
        .with_level(log::LevelFilter::Debug)
        .without_timestamps() // required for running wsl2
        .init()
        .unwrap();

    let client = ClientHandle::new("ws://localhost:9090").await?;
    let publisher = client.advertise::<std_msgs::Header>("talker").await?;

    loop {
        let msg = std_msgs::Header::default();
        info!("About to publish");
        publisher.publish(msg).await.unwrap();
        info!("Published msg...");
        tokio::time::sleep(tokio::time::Duration::from_secs(1)).await;
    }
}

```

ssnover, 2022-11-03:
Can you provide:
* `roslibrust` version you're using.
* Rust toolchain version.

It looks like the package and message you're using is this one right? https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/srv/DeviceInfo.srv 

I'd expect the generated output to look something like what you see for `diagnostics_msgs` in our example: https://github.com/Carter12s/roslibrust/blob/master/roslibrust_test/src/lib.rs#L72

```rust
#[allow(unused_imports)]
pub mod realsense2_camera {
    // ...
    pub struct DeviceInfoRequest {
        // members
    }
    pub struct DeviceInfoResponse {
        // members
    }
    // ...
}
```

The fact the message is reporting `realsesnse2_camera::DeviceInfo` implies to me that you may be on an older version that didn't support services perhaps? You'll also want to make sure that you import `serde` in your `Cargo.toml` which is generating the other error.

Akumar201, 2022-11-03:
Hi, thank you for your response, I am new to RUST, I want to create a publisher and send the data to another computer where ROS will be running. I tried creating my own package from the following example(https://github.com/Carter12s/roslibrust/blob/master/roslibrust/examples/basic_publisher.rs) . I am not using any realsense msg for this example, but I don't know why this error is associated with it. Thank you in advance. 

This is my cargo.toml file 

[package]
name = "ros_rust_bridge"
version = "0.1.0"
edition = "2021"

[dependencies]
roslibrust = "0.5.2"
tokio = "1.21.2"
rosrust_msg = "0.1.6"
log = "0.4.17"
simple_logger = "4.0.0"
roslibrust_codegen_macro = "0.5.2"
rosrust = "0.9"
serde = "1.0.147"

Carter12s, 2022-11-03:
Hey @Akumar201, thanks for trying out the library and filing the issue.

By default calling the `find_and_generate_ros_messages!()` macro will look in your ROS_PACKAGE_PATH environment variable, and generate messages for all ROS packages it can find in the paths listed there. In your case it does look like it might be finding multiple copies of the realsense2 package (this is likely a bug in library where we can make it resistant to this).

@ssnover and I can look into what happens if multiple versions of a package are found and try to make the library more resistant to this case.

It is also probably a good idea for us to offer a version of the macro that explicitly doesn't use ROS_PACKAGE_PATH. 

In the meantime you can work around this by un-setting the ROS_PACKAGE_PATH environment variable in the terminal you are building your package with, and then explicitly adding the paths you want to search for messages in into the macro.

If you are using bash as your terminal the command to unset ROS_PACKAGE_PATH would be `unset ROS_PACKAGE_PATH`.

And then you'd want to modify the macro line to look something like:
`roslibrust_codegen_macro::find_and_generate_ros_messages!("<Path to where std_msgs is installed>");`

If you want to read up on how ROS_PACKAGE_PATH is used / set there is some documentation [here](http://wiki.ros.org/ROS/EnvironmentVariables).

Carter12s, 2022-11-03:
I do also notice that in your Cargo.toml you have both "roslibrust" and "rosrust_msg" as dependencies. Fair warning that "roslibrust" and "rosrust_msg" aren't designed to work together, and your issue may be coming from trying to use both of them. This should technically be possible if namespaces are managed carefully, but the two crates aren't interoperable.

Akumar201, 2022-11-03:
I was wondering should I try removing realsense or try seeing if I have multiple version and remove other, should it work. I am new to rust so it would be great if you can provide me more detail in solution. Sorry for inconvenience.

Carter12s, 2022-11-03:
Try seeing if you do have multiple versions, and remove any duplicates.

Please also run `echo $ROS_PACKAGE_PATH` and post the result. It will help me diagnose the issue.

Carter12s, 2022-11-04:
@ssnover I attempted to replicate this issue on [this branch](https://github.com/Carter12s/roslibrust/tree/54-conflicting-implementation-for-realsense2_cameradeviceinfo) by creating duplicate copies of our test_msgs package and attempting generation.

Our de-duplication logic appears to be handling the case of multiple copies of a package perfectly:
```rust
pub fn find_and_generate_ros_messages(additional_search_paths: Vec<PathBuf>) -> TokenStream {
    let mut search_paths = utils::get_search_paths();
    search_paths.extend(additional_search_paths.into_iter());
    debug!(
        "Codegen is looking in following paths for files: {:?}",
        &search_paths
    );
    let mut packages = utils::crawl(search_paths);
    packages.dedup_by(|a, b| a.name == b.name);
```

I was however able to force a failure somewhat matching what Akumar201 reported by intentionally creating both a .msg file and a .srv file with the same name within a single package. This does result in two instances of the struct being defined and a compilation error. Looking through the actual librealsense2 package on github it doesn't look like they are doing that anywhere  so not sure if this is actually the same failure mode.

I investigated whether this is a real bug or not, but according to [this ticket](https://answers.ros.org/question/232050/name-conflict-between-msg-and-srv/) its illegal in ROS1 to have this name conflict. Regardless I think I might go ahead and make submodules in the generated packages to seperate out `msg` and `srv` since it looks like we'll have to for ROS2 support.

@Akumar201 I've not been able to reproduce your issue by my testing. So we will need more information from you to be able to help more.

Akumar201, 2022-11-05:
Hi, I resolved the issue, basically while downloading relasense sdk I downloaded two version of it's dependency libersense that was causing problem. I removed and install only latest version and this is working great. 

I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate

Akumar201, 2022-11-07:
Hi, I am reopening this issue as initially when I reinstall everything from scratch at least that code was not there. I though reason might be installing reaslsense2 two versions as suggested by you guys , however after a while the same error message is coming, I am not sure what is the problem with it. Is there any other method where I can use different message ?

ssnover, 2022-11-07:
Are you able to post your `ROS_PACKAGE_PATH` environment variable? i.e. wherever you're running `cargo build` can you run `echo $ROS_PACKAGE_PATH` and post the output here?

Alternatively, can you tell us your exact method of installing realsense2? This would make it easier for us to try to recreate your environment. Did you install with `apt` or perhaps cloning the repo into your catkin_ws? Different method?

Akumar201, 2022-11-07:
Hi, I am new to development. So, I would like to apologize for easy questions. 
```
echo $ROS_PACKAGE_PATH
/home/akumar/catkin_ws/src:/opt/ros/noetic/share
```
It worked with the method when I install realsense using the `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`, after then I did alot of installation like and upgrades like 
sudo apt update
sudo apt upgrade
pip install opencv-contrib-python
and other packages like moveit and transform3d 
I don't think these packages should have a effect on what we are getting.

Carter12s, 2023-02-15:
This is should be closed now that #81 has merged, next released version will contain this fix, master branch can be used in the meantime

closed 2023-02-15 by Carter12s