RosLibRust/roslibrust #57: WARN  [roslibrust::rosbridge::client] Failed to reconnect: CommFailure(Io(Os { code: 111, kind: ConnectionRefused, message: "Connection refused" }))

Akumar201, 2022-11-04:
Hi, I created my won package using  [example ](https://github.com/adnanademovic/rosrust/issues/179) . I am getting the following 

`[roslibrust::rosbridge::client] Failed to reconnect: CommFailure(Io(Os { code: 111, kind: ConnectionRefused, message: "Connection refused" }))`

Code I am using is 

```
use log::*;
use roslibrust::ClientHandle;

roslibrust_codegen_macro::find_and_generate_ros_messages!("assets/ros1_common_interfaces/std_msgs");

/// This example creates a client, and publishes a message to the topic "talker"
/// Running this example at the same time as subscribe_and_log will have the two examples
/// pass messages between each other.
/// To run this example a rosbridge websocket server should be running at the deafult port (9090).
#[tokio::main(flavor = "multi_thread")]
async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
    simple_logger::SimpleLogger::new()
        .with_level(log::LevelFilter::Debug)
        .without_timestamps() // required for running wsl2
        .init()
        .unwrap();

    let client = ClientHandle::new("ws://localhost:9080").await?;
    let publisher = client.advertise::<std_msgs::Header>("talker").await?;

    loop {
        let msg = std_msgs::Header::default();
        info!("About to publish");
        publisher.publish(msg).await.unwrap();
        info!("Published msg...");
        tokio::time::sleep(tokio::time::Duration::from_secs(1)).await;
    }
}

```

ssnover, 2022-11-04:
Are you running the rosbridge server on the host? You mentioned in the other issue that you were attempting to send a message to a remote machine, but I see the URI here includes `localhost` which will of course not be a remote machine.

Akumar201, 2022-11-05:
I am trying on the same machine then I will try sending it to remote, I would like to know is there any specific format in which I should enter the ip address of remote computer

Akumar201, 2022-11-08:
I found out the work around where I am using another system without realsense. It is acceptable in my work case. 
I figure out that It is essential to run rosbridge on the computer where you want to send data to. After installing rosbridge using the following command `sudo apt-get install ros-<rosdistro>-rosbridge-server` and launching using the command `roslaunch rosbridge_server rosbridge_websocket.launch` it was working perfectly. I would like to suggest to take care of the port number as rosbridge usually use 9090 and any different port number in the code could lead to error. 

I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2022-11-08 by Akumar201