RosLibRust/roslibrust #60: use of undeclared crate or module `std_msgs`

Akumar201, 2022-11-14:
Hi, I am using this package to send data from raspberrypi where I don't want to install ROS, however, I am not sure how can I define the std::msg by my own, I have seen some examples in https://github.com/Carter12s/roslibrust/tree/master/assets/test_msgs/msg . I also tried copying assert folder to my package. 
 
My code looks , something like this 

```
use log::*;
use roslibrust::ClientHandle;

roslibrust_codegen_macro::find_and_generate_ros_messages!("assets/ros1_common_interfaces/std_msgs");

/// This example creates a client, and publishes a message to the topic "talker"
/// Running this example at the same time as subscribe_and_log will have the two examples
/// pass messages between each other.
/// To run this example a rosbridge websocket server should be running at the deafult port (9090).
#[tokio::main(flavor = "multi_thread")]
async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
    simple_logger::SimpleLogger::new()
        .with_level(log::LevelFilter::Debug)
        .without_timestamps() // required for running wsl2
        .init()
        .unwrap();

    let client = ClientHandle::new("ws://192.169.XX.XX:9090").await?;
    let publisher = client.advertise::<std_msgs::Header>("talker").await?;

    loop {
        let msg = std_msgs::Header::default();
        info!("About to publish");
        publisher.publish(msg).await.unwrap();
        info!("Published msg...");
        tokio::time::sleep(tokio::time::Duration::from_secs(1)).await;
    }
}
```

Akumar201, 2022-11-14:
Hi, I resolved this issue by cloning the [std_msgs](https://github.com/ros/std_msgs) package and using by providing the path to the package. 

I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate

Moonpax, 2022-12-11:
Please show code example @Akumar201

Carter12s, 2022-12-12:
@Moonpax If you are looking for a solution here, I can provide some advice, and I'm also working on a documentation update (and hopefully a fundamentally better error message).

This failure will occur when you attempt message generation on a path (or set of paths) which don't contain all of the referenced packages within the messages.

So in this case, it is likely that the path passed into "find_and_generate_ros_messages" contained messages which relied on the std_msgs package. In this case roslibrust will generate Rust types for these messages, but they will be unable to compile because they are referencing other types that didn't get generated.

The easy solutions to this are:
1. Set ROS_PACKAGE_PATH ahead of compiling, and roslibrust will automatically look for additional packages there.
2. Add additional paths to the find_and_generate_ros_messages call that point to where std_msgs is installed on your system.

The error message that we should be giving the user in this case would be something like:

```
Error generating ros messages, cannot generate on file MyMsg.msg because this file references another package `std_msgs` on line 6: `header Header`, and the package `std_msgs` was not found in the search paths provide.
```

Akumar201, 2023-01-19:
@Moonpax Sorry for late reply.  You just need to download the message type you want to use in my case std message from the following link https://github.com/ros/std_msgs and then use the location of the std_msgs folder in the following line 
`
roslibrust_codegen_macro::find_and_generate_ros_messages!("user/folder1/location2/std_msgs");`

After this you can use this to send msgs without installing ROS. I hope this helps