RosLibRust/roslibrust #61: Use of std_msgs for rust "help: use struct literal syntax instead: `MultiArrayLayout {}`"

Akumar201, 2022-11-14:
Hi, I want to know is there any respository that I can use for std_msgs like https://github.com/ros/std_msgs. As I am trying to use this with the roslibrust package but, with data types like Float64MultiArray.msg I am getting the follwong error
```

  let msg = std_msgs::Float64MultiArray {layout: MultiArrayLayout, data: Vec};
   |                                                          ^^^^^^^^^^^^^^^^ help: use struct literal syntax instead: `MultiArrayLayout {}`

32 |         let msg = std_msgs::Float64MultiArray {layout: MultiArrayLayout {}, data: Vec {}};
   |                                                                                   ^^^^^^ expected struct `std::vec::Vec`, found struct `Vec`
   |

```

Carter12s, 2022-11-14:
Checkout the ros1 branch of std_msgs. This library does not currently
support ros2.

Carter12s, 2022-11-14:
Akumar, you seem to be incorrectly initializing a vector in Rust.

Please refer to the documentation for Vec: https://doc.rust-lang.org/std/vec/struct.Vec.html

Vec in Rust does not support brace initialization, and must either be initialized with Vec::new(), or using the vec![] macro.

This is not an issue with `roslibrust`, but fundamental syntax for the Rust language. 

`roslibrust` is simply generating corresponding Rust structs for the messages you feed into it. You then need to fill out all of the field of these structs yourself correctly when creating a message. Float64MultiArray is a complicated message type and I would not reccomend working with it until you are quite familiar with ROS itself:
ntation on all multiarrays.

ROS provides some documentation for what the various fields in this message should be here: http://docs.ros.org/en/melodic/api/std_msgs/html/msg/MultiArrayLayout.html

What you are likely trying to do is something like:

```rust
let msg = std_msgs::Float64MultiArray {
  layout: std_msgs::MultiArrayLayout {
    dim: vec![], // You actually need to fill this vec out for the message to work
    data_offset: 0, // You will need to calculate the data_offset and fill this out for the message to work
  }
  data: vec![] // your raw data will go here
}
```

RosLibRust does not (yet) provide a wrapper type for conveniently working with ROS multi-arrays (which are a weird and gross part of ROS in general). So, you are dealing with the raw message definition which requires knowledge of ROS (on top of understanding Rust syntax).

I recommend you work with a simpler message, until you are more comfortable with ROS, Rust, and `roslibrust`.

closed 2022-11-14 by Carter12s