RosLibRust/roslibrust #72: Stop leaking internal dependencies through generated code...

Carter12s, 2022-12-22:
Due to lack of use of $crate in the generated proc_macro code, crates using the proc_macro must themselves depend on serde (and now smart_default) to not have compiler errors.

We should fix this!

Carter12s, 2022-12-23:
@ssnover 

Someday can you take a look at:
https://github.com/Carter12s/roslibrust/tree/fix_dep_leak

I'm incredibly stuck... For some reason what I'm attempting to do works perfectly for smart_deafult, but not for serde...

I'm trying to follow: https://users.rust-lang.org/t/proc-macros-using-third-party-crate/42465/4 but clearly I'm missing something.

ssnover, 2023-02-04:
Looks like this is due to serde: https://github.com/serde-rs/serde/issues/1465

I think the fix mentioned in the thread will work, pushed it to the branch if you want to check it out.

PR #155 (2024-05-21): Fixing dependency leaks
## Description
Makes generated code only depend on types exposed via roslibrust_codgen so that users of that crate don't need to specify additional dependencies in an intuitive way.

## Fixes
Closes: #154
Closes: #72

## Checklist
- [ ] Update CHANGELOG.md
closed 2024-05-21 by Carter12s