RosLibRust/roslibrust #73: the name `ApplyPlanningScene` is defined multiple times `ApplyPlanningScene` must be defined only once in the type namespace of this modulerustcClick for full compiler diagnostic

Akumar201, 2023-01-19:
Hi, I am using roslibrust to send data from a system that is having ROS install with moveit and gazebo. Following is the example I am following, however, I am getting the following error

```
the name `ApplyPlanningScene` is defined multiple times
`ApplyPlanningScene` must be defined only once in the type namespace of this modulerustc[Click for full compiler diagnostic](rust-analyzer-diagnostics-view:/diagnostic%20message%20%5B1%5D?1#file%3A%2F%2F%2Fhome%2Fakumar%2FTargetArm_Project%2Ftargetarm%2Fmessage_broadcaster%2Fsrc%2Fmain.rs)
the name `ChangeControlDimensions` is defined multiple times
`ChangeControlDimensions` must be defined only once in the type namespace of this modulerustc[Click for full compiler diagnostic](rust-analyzer-diagnostics-view:/diagnostic%20message%20%5B2%5D?2#file%3A%2F%2F%2Fhome%2Fakumar%2FTargetArm_Project%2Ftargetarm%2Fmessage_broadcaster%2Fsrc%2Fmain.rs)
the name `ChangeDriftDimensions` is defined multiple times
`ChangeDriftDimensions` must be defined only once in the type namespace of this modulerustc[Click for full compiler diagnostic](rust-analyzer-diagnostics-view:/diagnostic%20message%20%5B3%5D?3#file%3A%2F%2F%2Fhome%2Fakumar%2FTargetArm_Project%2Ftargetarm%2Fmessage_broadcaster%2Fsrc%2Fmain.rs)

```
```
use log::*;
use roslibrust::ClientHandle;

roslibrust_codegen_macro::find_and_generate_ros_messages!("/opt/ros/noetic/share/std_msgs");

#[tokio::main(flavor = "multi_thread")]
async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
    simple_logger::SimpleLogger::new()
        .with_level(log::LevelFilter::Debug)
        .without_timestamps() // required for running wsl2
        .init()
        .unwrap();
```

    let client = ClientHandle::new("ws://localhost:9090").await?;
    let publisher = client.advertise::<std_msgs::String>("talker").await?;
    let val = "Hello World"; // sending example string to ros node working on another computer 
    loop {
        let msg = std_msgs::String {data: val.to_string()};
        info!("About to publish");
        publisher.publish(msg).await.unwrap();
        info!("Published msg...");
        tokio::time::sleep(tokio::time::Duration::from_secs(1)).await;
    }
}

ssnover, 2023-02-04:
Hey @Akumar201 I've been working on some better logs for the code generation, although it requires using the runtime method. In order to debug this I need you to try out my open branch: `git clone -b better-logs-for-duplicate-msgs https://github.com/Carter12s/roslibrust.git`.

Then you can modify the paths under `roslibrust/roslibrust_test/src/main.rs` in the variable `ROS_1_PATHS` to be: 
```rs
static ref ROS_1_PATHS: Vec<PathBuf> = vec!["/opt/ros/noetic/share/std_msgs".into()];
```

Finally, run with `RUST_LOG=debug cargo run --bin roslibrust_test` and you'll be able to see where the library is discovering packages, and if we've found any duplicates. This will help us narrow down the problem. We do look for duplicates currently, but somehow your environment is defeating the mechanism (we de-duplicate by package name).

Carter12s, 2023-02-15:
Should be resolved by #81 next released version will contain fix, master branch can be used for now.

closed 2023-02-15 by Carter12s