RosLibRust/roslibrust #79: ROS1: Work needed to support native ROS1 nodes

ssnover, 2023-02-04:
Did a little spelunking through ros/ros_comm some time ago and I think the things needed to achieve basic pubsub on ROS1 are:
* Start generating md5sums for message types and add a method to get that md5sum
* For publishers:
  * Host an XMLRPC on a TCP port
  * Call `registerPublisher` method on rosmaster XMLRPC API.
  * Handle connection headers from subscribers which want to listen by responding with header response
  * Then send rosmsg encoded data on the stream above
* For subscribers
  * Request the XMLRPC URIs from rosmaster for a given topic
  * Listen for new publishers registered on an XMLRPC URI
  * Make TCPROS connections to publishers (first with a connection header), wait for connection header response
  * Handle incoming rosmsg encoded data.

Relevant docs:
* [TCPROS docs](http://wiki.ros.org/ROS/TCPROS)
* [rosmsg wire format](http://wiki.ros.org/ROS/Connection%20Header)

ssnover, 2023-02-04:
Honestly, the docs are pretty lackluster, I'll go dig up the packet capture I took, attach to here, and then write up more complete information.

Carter12s, 2024-05-24:
We're not 100% done with ROS1 support, but don't think this issue is super useful for tracking anymore.

Publisher, Subscribers, and Service Clients are all working in basic form.

closed 2024-05-24 by Carter12s