RosLibRust/roslibrust #82: Duplicate service struct definitions are constructed if the same service appears more than once

ssnover, 2023-02-10:
Copy the contents of `assets/ros1_common_interfaces/common_msgs/diagnostic_msgs` to another directory, maybe `/tmp`. Add this to your search paths somehow (`ROS_PACKAGE_PATH=/tmp`) and run `RUST_LOG=debug cargo run --bin roslibrust_test`.

The conflict is correctly noticed and msg files are not duplicated. However, service files are duplicated! Given the example, search `roslibrust_test/src/ros1.rs` and you'll find two entries for the `AddDiagnostic` service!

Carter12s, 2023-02-10:
Heck yeah. Damn good find. Should be an easy fix now that we've spotted it.

ssnover, 2023-02-10:
The fact I managed to trigger it by modifying the CI environment is nice because we automatically start testing for it in CI now.

Carter12s, 2023-02-15:
Closed in #81

closed 2023-02-15 by Carter12s