RosLibRust/roslibrust #108: Support for passing async fn into `advertise_service`

entire, 2023-08-23:
In the `service_server.rs` example, I've noticed that there is:

```rust
    let _handle = client
        .advertise_service::<std_srvs::SetBool>("/my_set_bool", my_service)
        .await?;
```
and `my_service` is a simple `fn` here, but many services are often going to require async/await type functions to run within the function. Is there plan to support async fn here? If not, I'm happy to look into creating a PR to add support for this especially because I'd like to use async functions within my service callback that I pass in.

ssnover, 2023-08-23:
If you're blocked on this you still have the option of calling `tokio::spawn` here.

entire, 2023-08-24:
@ssnover agreed - i think it would just be nicer overall though for the functions to be async in general though.

@Carter12s - thanks, makes sense. let me take a look at it and I'll take a stab at it if it seems relatively not difficult.

ssnover, 2023-09-07:
Having glanced at this for a bit, the main challenge is our paradigm around type erasure with the closure. Async closures are unstable, which means the `message: &str` parameter will have to find another way in. 

If you look at implementing this: I'd recommend `tokio::spawn` with a channel. Wrap the `tokio::JoinHandle` in an `abort_on_drop::ChildTask` and add that to the struct that gets store in the services map along with the sender side of the channel for messages.

Carter12s, 2024-12-17:
Update!

Async closures have merged! https://github.com/rust-lang/rust/pull/132706

We should be able to build this out once the next version of rust ships (could build it out on nightly right now).

Once built we can put it behind a feature flag to not affect out min rust version too much. But super excited this is coming.

Carter12s, 2025-02-21:
Rust 1.85 is officially out with async closures are fully supported!!!

We're ready to start work on this one.

ssnover, 2025-02-21:
Do we have an established MSRV target? It's pretty common for the ecosystem to support current stable and the two previous versions.

Carter12s, 2025-02-21:
README currently calls out 1.75, but I don't think we've got that in the Cargo.tomls yet.

My thought here is that we build this feature out under a feature flag ["async-closures"], and when ~1.87 drops we enable that feature as default.

I'm open to opinions on the API, but it feels like it is worth keeping the non-async versions around? Have both `advertise_service` and `advertise_async_service`? That makes the change non-breaking, additive, and pretty clean behind a feature flag.

Carter12s, 2025-03-31:
Okay I took a stab at building this out (#227), but did NOT have a lot of luck. Ran into the following issues:

1. `AsyncFn` is not dyn compatible, so storing a list of `AsyncFn`s in some kind of data structure ends up being a massive pain in the ass. Instead you have to do a bunch of shenanigans to hide the async closures in a manually created closure that returns a future.  This made writing the feature very challenging, but was surmountable.
2. An async closure which captures ANY context by move is not `Fn` and is instead `FnOnce`. This is because multiple futures can be spawned from the async closure in parallel and they can't all have ownership over the moved context at the same time. These leads to fairly horrible ergonomics for users of the feature.

Where you'd like to write:

```rust
         let service = async move |request: std_srvs::SetBoolRequest| -> std::result::Result<
             std_srvs::SetBoolResponse,
             Box<dyn std::error::Error + Send + Sync>,
         > {
             tx.send(request.data).await.unwrap();
             Ok(std_srvs::SetBoolResponse {
                 success: true,
                 message: "You set my bool!".to_string(),
             })
         };
```

You instead have to write:

```rust
        let service = move |request: std_srvs::SetBoolRequest| {
            let tx = tx.clone();
            Box::pin(async move {
                tx.send(request.data).await.unwrap();
                Ok(std_srvs::SetBoolResponse {
                    success: true,
                    message: "You set my bool!".to_string(),
                })
            })
        };
```

I may have totally screwed something up in the implementation, so would love other eyes on my prototype. But, based on how this initial try wen't I'm not sure I want to move forward with this.

Carter12s, 2025-03-31:
^Yeah I think an API that provides a runtime handle, or frankly an API which sets up a pair of channels and makes it easy for the user to spawn a tokio::task() to be the service are both more likely to end up ergonomic than actually taking in an async closure...

Would be worth doing a bit of an API study and seeing what other systems have come up with.

Carter12s, 2025-04-07:
Okay I took a second swing at this based and ultimately decided that having a single `advertise_service` function was the best API.

Inside the user's service callback they can use `tokio::Handle::current().block_on` to do whatever async stuff they need, and this gives much more clear and understandable compiler errors to the user. Everything with `AsyncFn` just led to very arcane and non-obvious error messages.

To support this change all service functions need to be executing inside a `tokio::task::spawn_blocking` which was probably the better design call from the get-go.

I'd like to continue to improve documentation for this, but I think having an example and some bread-crumbs is good enough for now.

closed 2025-04-07 by Carter12s

Carter12s, 2025-04-07:
Roslibrust v0.14.0 published with examples on how to do this.

PR #228 (2025-04-07): Async Service Examples and Backend Fixes
## Description
- Changes all back-ends to always call user provided service functions inside of `spawn_blocking`
- Adds example showing how to perform async operations inside of a service handle
- Adds integration test confirming this behavior for ros1

## Fixes
Closes: #108 

## Checklist
- [x] Update CHANGELOG.md