RosLibRust/roslibrust #124: Separate traits for RosMessageType, MD5SSUM, and Source?

Carter12s, 2023-09-28:
So one cool ability of roslibrust I'm planning to highlight in my talk was the ability to hand define a generic version of message, and leverage serde_json to correct deserialize the right version of it at runtime. Going to be filing a PR to add an example of this soon (today hopefully).

Ran into an issue while doing this that we introduced on the way to ROS1 native support.

We extended the requirements to be a RosMessageType to include 

```
    /// The computed md5sum of the message file and its dependencies
    const MD5SUM: &'static str;

    /// The definition from the msg, srv, or action file
    const DEFINITION: &'static str;
```

But these fields aren't actually required everywhere, and they make it way harder / confusing to hand define a generic type. I think we should move these to their own trait, which is only sometimes required depending on what you are using the message for?

ssnover, 2023-09-28:
Maybe we could just add defaults to those two fields? While this use case is interesting, I wouldn't expect it to be nominal or common and I'm worried about adding complexity to accommodate it.

Carter12s, 2023-09-28:
If we can provide defaults of empty string "" I think that is more than good enough. We could then provide runtime errors if you do end up calling a function that needs it to not be ""? I didn't know defaults were really possible! Just read the documentation on them and that seems like a nice "good enough" for now. I'll file an MR to add defaults

ssnover, 2023-09-28:
I think that the only thing which might require this is native ROS connections, in which case if you're defining messages by hand I'd call that at-your-own-risk behavior and let them debug from errors issued by rosmaster, failures to match md5sums with publisher nodes, etc.

closed 2023-10-03 by Carter12s