RosLibRust/roslibrust #128: Codegen: dependency iteration limit

NathanSWard, 2023-09-29:
Currently running into the iteration limit for codegen: 
```
ERROR roslibrust_codegen: Unable to resolve dependencies after reaching iteration limit (2048)
```
Would it be reasonable to have a feature flag to disable the limit or some other way to specify the limit?

Carter12s, 2023-09-29:
Hey Nathan, I'll probably take some time to improve the error message right away, because it is highly likely you have a different problem then that message actually points you at.

That iteration limit would only be hit if you had a scenario where:

Message A Needs -> Message B Needs -> Message C ........... All the way to Message XXXZZX or whatever. You would have to have a dependency chain more that 2048 message long.

Instead what is likely going on here is that you pointed the codegen at some directories where it found messages that referenced other messages in other packages, but those packages weren't found in any of its search paths.

That message should really say something like:

"After reaching the maximum iteration limit for resolving dependencies failed to find message definition for {missing_message} which is needed by {message_with_dependency}, consider adding the path for the package containing that message to your search paths, or to the ROS_PACKAGE_PATH environment variable"

I'm actually doing an overhaul of the error messages produced by codegen at the moment, and will work on that shortly!

Carter12s, 2023-09-30:
PR #127 is up and significantly improves this error message into something like:

[image]

closed 2023-10-03 by ssnover

Carter12s, 2023-10-04:
@NathanSWard v0.8 of roslibrust, roslibrust_codegen, and roslibrust_codegen_macro were released moments ago.

Would be very curious to see what errors you get if you re-attempt what you were trying to do with those versions.

NathanSWard, 2023-10-05:
Thanks @Carter12s !
I'll give it a spin this weekend and get back to you on if this helped with the errors messages!
Appreciate the quick turn around :D