RosLibRust/roslibrust #141: What is going on with doctest paths in Cargo

Carter12s, 2023-10-05:
@ssnover I need a exorcist...

I can't for the life of me figure out what is going on here.

To reproduce (which our CI is doing) use rust 1.72:

Run `cargo test`

Confirm these lines present in output:
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::advertise (line 292) - compile ... ok
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::subscribe (line 201) - compile ... ok
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle (line 54) - compile ... ok
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::call_service (line 344) - compile ... ok
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::subscribe (line 226) - compile ... ok

Run `cargo test --features rosapi`
You see this instead:
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::subscribe (line 201) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::advertise (line 292) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::call_service (line 344) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle (line 54) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::subscribe (line 226) - compile ... FAILED

With error messages of:
---- roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::advertise (line 292) stdout ----
error: Codegen was instructed to search a path that could not be canonicalized relative to "/home/carter/roslibrust/roslibrust_codegen_macro": "assets/ros1_common_interfaces/std_msgs", No such file or directory (os error 2)
 --> roslibrust/src/rosbridge/client.rs:294:1
  |
4 | / roslibrust_codegen_macro::find_and_generate_ros_messages!(
5 | |    "assets/ros1_common_interfaces/std_msgs"
6 | | );
  | |_^
  |
  = note: this error originates in the macro `roslibrust_codegen_macro::find_and_generate_ros_messages` (in Nightly builds, run with -Z macro-backtrace for more info)
  
SOMEHOW SETTING THE `rosapi` FEATURE FLAG IS CHANGING THE WORKING DIRECTORY FOR THE DOC TESTS

ssnover, 2023-10-05:
Have you attempted writing out the working directory in the error message to confirm?

ssnover, 2023-10-05:
Oh I guess that's the first path.

Carter12s, 2023-10-06:
Plot thickens. The actual command being invoked has `--test-run-directory of roslibrust` set correctly in the actual invocation of rustdoc...

```
/home/carter/.rustup/toolchains/1.72-x86_64-unknown-linux-gnu/bin/rustdoc --edition=2021 --crate-type lib --crate-name roslibrust --test roslibrust/src/lib.rs --test-run-directory /home/carter/open_source/roslibrust/roslibrust -L dependency=/home/carter/open_source/roslibrust/target/debug/deps -L dependency=/home/carter/open_source/roslibrust/target/debug/deps --extern abort_on_drop=/home/carter/open_source/roslibrust/target/debug/deps/libabort_on_drop-ebe36dd30e31a7da.rlib --extern anyhow=/home/carter/open_source/roslibrust/target/debug/deps/libanyhow-47cda635cc19f115.rlib --extern async_trait=/home/carter/open_source/roslibrust/target/debug/deps/libasync_trait-e8a0058d4a061a60.so --extern byteorder=/home/carter/open_source/roslibrust/target/debug/deps/libbyteorder-4c38abb7151e1d8f.rlib --extern dashmap=/home/carter/open_source/roslibrust/target/debug/deps/libdashmap-9e16df30a730a27b.rlib --extern deadqueue=/home/carter/open_source/roslibrust/target/debug/deps/libdeadqueue-5d89cd2a7ebc7d2a.rlib --extern env_logger=/home/carter/open_source/roslibrust/target/debug/deps/libenv_logger-fbd7a842b22eef13.rlib --extern futures=/home/carter/open_source/roslibrust/target/debug/deps/libfutures-017373bc813ea370.rlib --extern futures_util=/home/carter/open_source/roslibrust/target/debug/deps/libfutures_util-10817f2ccb5f296d.rlib --extern lazy_static=/home/carter/open_source/roslibrust/target/debug/deps/liblazy_static-ea967ec45ae077a3.rlib --extern log=/home/carter/open_source/roslibrust/target/debug/deps/liblog-439ed6363b15720a.rlib --extern proc_macro2=/home/carter/open_source/roslibrust/target/debug/deps/libproc_macro2-1d966d3deb3e4d76.rlib --extern rand=/home/carter/open_source/roslibrust/target/debug/deps/librand-79d7f7f977f40bf6.rlib --extern roslibrust=/home/carter/open_source/roslibrust/target/debug/deps/libroslibrust-54977813f42d1ed1.rlib --extern roslibrust_codegen=/home/carter/open_source/roslibrust/target/debug/deps/libroslibrust_codegen-3d2c177fe306cb6d.rlib --extern roslibrust_codegen_macro=/home/carter/open_source/roslibrust/target/debug/deps/libroslibrust_codegen_macro-2c424e09d3e3f215.so --extern serde=/home/carter/open_source/roslibrust/target/debug/deps/libserde-e33ac9ddd7cb1512.rlib --extern serde_json=/home/carter/open_source/roslibrust/target/debug/deps/libserde_json-3fa0ed1f2e00fdd9.rlib --extern simple_logger=/home/carter/open_source/roslibrust/target/debug/deps/libsimple_logger-90511ecaa141f13a.rlib --extern smart_default=/home/carter/open_source/roslibrust/target/debug/deps/libsmart_default-141bc7be9eef9078.so --extern test_log=/home/carter/open_source/roslibrust/target/debug/deps/libtest_log-0f4daa73fc3ece64.so --extern thiserror=/home/carter/open_source/roslibrust/target/debug/deps/libthiserror-5f1822288c444960.rlib --extern tokio=/home/carter/open_source/roslibrust/target/debug/deps/libtokio-fc4055114e1160fd.rlib --extern tokio_tungstenite=/home/carter/open_source/roslibrust/target/debug/deps/libtokio_tungstenite-bd38240000129bbd.rlib --extern uuid=/home/carter/open_source/roslibrust/target/debug/deps/libuuid-3fa9c524d020759a.rlib -C embed-bitcode=no --cfg 'feature="default"' --cfg 'feature="rosapi"' --error-format human`

running 8 tests
test roslibrust/src/lib.rs - (line 36) ... ignored
test roslibrust/src/lib.rs - (line 41) ... ignored
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::advertise (line 292) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle (line 54) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::subscribe (line 201) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::call_service (line 344) - compile ... FAILED
test roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::subscribe (line 226) - compile ... FAILED
test roslibrust/src/lib.rs - (line 48) ... ok

failures:

---- roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::advertise (line 292) stdout ----
error: Codegen was instructed to search a path that could not be canonicalized relative to "/home/carter/open_source/roslibrust/roslibrust_codegen_macro": "assets/ros1_common_interfaces/std_msgs", No such file or directory (os error 2)
 --> roslibrust/src/rosbridge/client.rs:294:1
 ```

Carter12s, 2023-10-06:
Working command appears identical:

```
     Running `/home/carter/.rustup/toolchains/1.72-x86_64-unknown-linux-gnu/bin/rustdoc --edition=2021 --crate-type lib --crate-name roslibrust --test roslibrust/src/lib.rs --test-run-directory /home/carter/open_source/roslibrust/roslibrust -L dependency=/home/carter/open_source/roslibrust/target/debug/deps -L dependency=/home/carter/open_source/roslibrust/target/debug/deps --extern abort_on_drop=/home/carter/open_source/roslibrust/target/debug/deps/libabort_on_drop-ebe36dd30e31a7da.rlib --extern anyhow=/home/carter/open_source/roslibrust/target/debug/deps/libanyhow-47cda635cc19f115.rlib --extern async_trait=/home/carter/open_source/roslibrust/target/debug/deps/libasync_trait-e8a0058d4a061a60.so --extern byteorder=/home/carter/open_source/roslibrust/target/debug/deps/libbyteorder-4c38abb7151e1d8f.rlib --extern dashmap=/home/carter/open_source/roslibrust/target/debug/deps/libdashmap-9e16df30a730a27b.rlib --extern deadqueue=/home/carter/open_source/roslibrust/target/debug/deps/libdeadqueue-5d89cd2a7ebc7d2a.rlib --extern env_logger=/home/carter/open_source/roslibrust/target/debug/deps/libenv_logger-fbd7a842b22eef13.rlib --extern futures=/home/carter/open_source/roslibrust/target/debug/deps/libfutures-017373bc813ea370.rlib --extern futures_util=/home/carter/open_source/roslibrust/target/debug/deps/libfutures_util-10817f2ccb5f296d.rlib --extern lazy_static=/home/carter/open_source/roslibrust/target/debug/deps/liblazy_static-ea967ec45ae077a3.rlib --extern log=/home/carter/open_source/roslibrust/target/debug/deps/liblog-439ed6363b15720a.rlib --extern proc_macro2=/home/carter/open_source/roslibrust/target/debug/deps/libproc_macro2-1d966d3deb3e4d76.rlib --extern rand=/home/carter/open_source/roslibrust/target/debug/deps/librand-79d7f7f977f40bf6.rlib --extern roslibrust=/home/carter/open_source/roslibrust/target/debug/deps/libroslibrust-4b49c85c1cdfad1e.rlib --extern roslibrust_codegen=/home/carter/open_source/roslibrust/target/debug/deps/libroslibrust_codegen-3d2c177fe306cb6d.rlib --extern roslibrust_codegen_macro=/home/carter/open_source/roslibrust/target/debug/deps/libroslibrust_codegen_macro-2c424e09d3e3f215.so --extern serde=/home/carter/open_source/roslibrust/target/debug/deps/libserde-e33ac9ddd7cb1512.rlib --extern serde_json=/home/carter/open_source/roslibrust/target/debug/deps/libserde_json-3fa0ed1f2e00fdd9.rlib --extern simple_logger=/home/carter/open_source/roslibrust/target/debug/deps/libsimple_logger-90511ecaa141f13a.rlib --extern smart_default=/home/carter/open_source/roslibrust/target/debug/deps/libsmart_default-141bc7be9eef9078.so --extern test_log=/home/carter/open_source/roslibrust/target/debug/deps/libtest_log-0f4daa73fc3ece64.so --extern thiserror=/home/carter/open_source/roslibrust/target/debug/deps/libthiserror-5f1822288c444960.rlib --extern tokio=/home/carter/open_source/roslibrust/target/debug/deps/libtokio-fc4055114e1160fd.rlib --extern tokio_tungstenite=/home/carter/open_source/roslibrust/target/debug/deps/libtokio_tungstenite-bd38240000129bbd.rlib --extern uuid=/home/carter/open_source/roslibrust/target/debug/deps/libuuid-3fa9c524d020759a.rlib -C embed-bitcode=no --cfg 'feature="default"' --error-format human`
```
[image]

Carter12s, 2023-10-06:
OMG THIS IS INSANE
Running the command individually on the command line the one on the right in the above image works. The one on the left does not...

Carter12s, 2023-10-06:
Have re-created the issue in a minimum viable example (well its pretty small still not sure 100% of it is needed).
https://github.com/Carter12s/bug-repo

Carter12s, 2023-10-06:
I decided to file a compiler bug:
https://github.com/rust-lang/rust/issues/116495

ssnover, 2023-10-07:
I've recreated on Rust 1.73. Getting errors with `cargo test --features rosapi`:
```
---- roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::call_service (line 344) stdout ----
error: Codegen was instructed to search a path that could not be canonicalized relative to "/home/ssnover/dev/roslibrust/roslibrust_codegen_macro": "assets/ros1_common_interfaces/rosapi", No such file or directory (os error 2)
```

It appears to be taking the current working directory and attaching the invoking proc-macro directory? In CI (https://github.com/Carter12s/roslibrust/actions/runs/6422967246/job/17440516311?pr=139):
```
Couldn't compile the test.
---- roslibrust/src/rosbridge/client.rs - rosbridge::client::ClientHandle::advertise (line 292) stdout ----
error: Codegen was instructed to search a path that could not be canonicalized relative to "/__w/roslibrust/roslibrust/roslibrust_codegen_macro": "assets/ros1_common_interfaces/std_msgs", No such file or directory (os error 2)
```

As far as I can tell this only affects doctests. It also does not happen when using `cargo build --tests --features rosapi`. That seems to indicate that the doctests aren't built by that invocation since this is of course happening at compile time.

Finally, if you run `cargo test --lib --bins --tests` the invocation runs without error which indicates that it is only the doctests that are causing issues.

ssnover, 2023-10-07:
Hmm so this does not currently occur on `5b18321`, but does occur on current `HEAD` of master; `dffd3a0`. Going to run through the history and find the offender.

ssnover, 2023-10-07:
Somehow, the offending commit appears to be:
`6051a2e Update readme`
Which is especially confusing because that commit only changes a markdown file:
```diff
git diff -w HEAD~1
diff --git a/README.md b/README.md
index 740b159..087c502 100644
--- a/README.md
+++ b/README.md
@@ -59,6 +59,10 @@ Rough overview of the features planned to built for this crate in what order:
 
 Contribution through reporting of issues encountered and implementation in PRs is welcome! Before landing a large PR with lots of code implemented, please open an issue if there isn't a relevant one already available and chat with a maintainer to make sure the design fits well with all supported platforms and any in-progress implementation efforts.
 
+### Rust Version for Development
+
+Due to cargo 1.72 enabling "doctest-in-workspace" by default it is reccomended to use Rust 1.72+ for develop. Previous rust versions are support but will require some incantations when executing doctests.
+
 ### Running Tests
 
 There are various unit tests and integration tests behind feature flags. For tests with ROS1, both through rosbridge and native clients, you'll need a locally running `rosbridge_websocket` node and `rosmaster`. Then run with `cargo test --features "ros1_test ros1"`. For tests with ROS2, you'll need a running rosbridge server, then run with `cargo test --features "ros2_test"`. You can find relevant `Dockerfile`s and docker compose configurations udner the `docker` directory.
```

ssnover, 2023-10-07:
Confirmed with Rust 1.71.1 so it doesn't appear to be `doctest-in-workspace` related at least at a surface level.

ssnover, 2023-10-07:
Oh I think maybe I scrolled which caused my git history in the scrollback buffer to get distorted before I copied it.

ssnover, 2023-10-07:
Yeah, okay, this is a smoking gun if I've ever seen one: 
```
ssnover@pop-os:~/dev/roslibrust$ rustc --version
rustc 1.73.0 (cc66ad468 2023-10-03)
ssnover@pop-os:~/dev/roslibrust$ git diff -w e2b64a3 6051a2e roslibrust/src/rosbridge/client.rs
diff --git a/roslibrust/src/rosbridge/client.rs b/roslibrust/src/rosbridge/client.rs
index 55832d3..d391d33 100644
--- a/roslibrust/src/rosbridge/client.rs
+++ b/roslibrust/src/rosbridge/client.rs
@@ -53,7 +53,7 @@ impl ClientHandleOptions {
 /// ClientHandle is clone and multiple handles can be clone()'d from the original and passed throughout your application.
 /// ```no_run
 /// # roslibrust_codegen_macro::find_and_generate_ros_messages!(
-/// #    "../assets/ros1_common_interfaces/std_msgs"
+/// #    "assets/ros1_common_interfaces/std_msgs"
 /// # );
 /// # #[tokio::main]
 /// # async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
@@ -200,7 +200,7 @@ impl ClientHandle {
     /// Subscribe to a given topic expecting msgs of provided type.
     /// ```no_run
     /// # roslibrust_codegen_macro::find_and_generate_ros_messages!(
-    /// #    "../assets/ros1_common_interfaces/std_msgs"
+    /// #    "assets/ros1_common_interfaces/std_msgs"
     /// # );
     /// # #[tokio::main]
     /// # async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
@@ -227,12 +227,12 @@ impl ClientHandle {
     /// # // TODO figure out how to de-duplicate code here with this message definition...
     /// # mod ros1 {
     /// # roslibrust_codegen_macro::find_and_generate_ros_messages!(
-    /// #   "../assets/ros1_common_interfaces/std_msgs"
+    /// #   "assets/ros1_common_interfaces/std_msgs"
     /// # );
     /// # }
     /// # mod ros2 {
     /// # roslibrust_codegen_macro::find_and_generate_ros_messages!(
-    /// #   "../assets/ros1_common_interfaces/std_msgs"
+    /// #   "assets/ros1_common_interfaces/std_msgs"
     /// # );
     /// # }
     /// # #[tokio::main]
@@ -291,7 +291,7 @@ impl ClientHandle {
     ///
     /// ```no_run
     /// # roslibrust_codegen_macro::find_and_generate_ros_messages!(
-    /// #    "../assets/ros1_common_interfaces/std_msgs"
+    /// #    "assets/ros1_common_interfaces/std_msgs"
     /// # );
     /// # #[tokio::main]
     /// # async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
@@ -343,7 +343,7 @@ impl ClientHandle {
     ///
     /// ```no_run
     /// # roslibrust_codegen_macro::find_and_generate_ros_messages!(
-    /// #    "../assets/ros1_common_interfaces/rosapi"
+    /// #    "assets/ros1_common_interfaces/rosapi"
     /// # );
     /// # #[tokio::main]
     /// # async fn main() -> Result<(), Box<dyn std::error::Error + Send + Sync>> {
```

Naturally, with the `../` prefix the tests pass.

Carter12s, 2023-10-08:
^Sorry I just want to be super clear. I'm very aware of that commit and made it originally to resolve the issue.

The actual issue I'm seeing is that SOMETIMES the "../" is required for the tests to pass and SOMETIMES the "../" will cause the tests to fail depending on seemingly random changes I make.

On all three of my systems right now:

- With the "../" removed all doc tests pass when running `cargo test`
- With the "../" removed those specific doc tests fail when running `cargo test --features rosapi`
- The inverse is true if the "../" is present
- On rust 1.71 the tests always pass with the "../" present, and fail with the "../" removed

I have fairly concrete evidence that on 1.72 the doctests are being **COMPILED** with the compilation working directory set the workspace root when no feature flag is set, and the working directory set to WORKSPACE_ROOT/roslibrust_codegen_macro when the "--features rosapi" flag is set. This is critical because the compilation directory is what is found when the proc_macro is expanded.

Right now adding the "../" back in causes CI to pass, but only because CI is invoking with the "--features rosapi" included. I'd be VERY curious to hear if you have different behavior on your system.

Some notes:
* I don't believe "--feature rosapi" is in anyway specifically at fault. I think that flag is simply causing tests / compilations to happen in a different order in such a way that issue manifests

Carter12s, 2023-10-08:
OH SHIT I THINK THAT WAS THE CLUE!!!

FUCK

I think this is caused by the macro "find_and_generate_ros_messages_relative_to_manifest_dir!" in WHICH I LIKE SOME KIND OF IDIOT CALL std::env:set_current_dir!!! Which probably takes affect for the rest of that compilation unit / job / file / whatever...

I'd bet money the root problem is that SOMETIMES within a single "instance" that macro is being invoked before the doctests, and SOMETIMES not being invoked before the doctests which causes the weird effects. I don't have time to investigate tonight, but I bet removing / replacing the function, or simply have it un-set the working dir after its bullshit fixes all this...

ssnover, 2023-10-08:
I would offer that it might be cargo's responsibility to limit environment changes caused by macros like this, although we definitely shouldn't be overwriting the working directory for any reason. Wasm proc macros can't come soon enough.

Carter12s, 2023-10-08:
Yeah I might open an issue to that affect when I can distill things down to a super clear example.

Working on actual fix as we speak. Double confirmation that set_current_dir was the problem, as the rosapi module was the place that that macro variant was being used.

Carter12s, 2023-10-08:
Also past me was right, he just didn't know how right:
[image]

PR #140 (2023-10-24): Trying to figure out which tests are flaky and fix, adding debug info…
## Description
Trying to make CI more robust

## Fixes
Closes: #122
Closes: #141

## Checklist
- [ ] Update CHANGELOG.md
closed 2023-10-13 by Carter12s