RosLibRust/roslibrust #178: PointCloud2 messages exceeding serde_rosmsg buffer size

lucasw, 2024-07-30:
I'm receiving progressively larger PointCloud2 generated messages from https://github.com/lucasw/transform_point_cloud/blob/master/scripts/generate_point_cloud.py but once they get too big they don't get successfully decoded in my [subscriber](https://github.com/lucasw/transform_point_cloud/blob/master/src/bin/transform_point_cloud2.rs#L71) (which is using the latest roslibrust commit in master here), and I see:

```
serde_rosmsg Error: Reached end of memory buffer while reading data
serde_rosmsg Error: Attempted to read beyond the end of decoded value's length
```

Maybe this code is involved - the 4096 size limit is in the range where I see the issue (but isn't it supposed to keep reading and assemble a full message?):

https://github.com/Carter12s/roslibrust/blob/master/roslibrust/src/ros1/subscriber.rs#L119-L120

Carter12s, 2024-07-30:
Whoops yup that is the smoking gun.

I identified this issue when implementing the logic for service servers and fixed it correctly there:
https://github.com/Carter12s/roslibrust/blob/8e0066cbc2370ab321c7bac333f0748bf2d0ef4b/roslibrust/src/ros1/service_server.rs#L216

But failed to port that fix back to subscriber :facepalm:

Will patch shortly

Carter12s, 2024-07-30:
Fix is merged to master

Will publish v0.10.2 with fix to crates.io shortly

PR #179 (2024-07-30): Fix 178 subscriber buffer size
## Description
- Fixed bug where subscribers weren't handling message size correctly
- Standardized logic for receiving a ROS1 message body that was duplicated in subscriber, service_client and service_server
- Added a unit test for subscribers for large payload

## Fixes
Closes: #178 

## Checklist
- [x] Update CHANGELOG.md
closed 2024-07-30 by Carter12s