RosLibRust/roslibrust #182: Publisher latching

lucasw, 2024-08-02:
I see a few instances of `latching` in the code but there isn't logic to actually publish a latched message, it only sets the flag in the connection header so far?

It looks like it would be a modification to the listener handle https://github.com/Carter12s/roslibrust/blob/master/roslibrust/src/ros1/publisher.rs#L91 to support, have hold onto the last sent message and publish it like in rospy topics.py: https://github.com/strawlab/ros_comm/blob/master/clients/rospy/src/rospy/topics.py#L809-L811 and https://github.com/strawlab/ros_comm/blob/master/clients/rospy/src/rospy/topics.py#L851-L852

I've been trying out dynamic reconfigure [here](https://github.com/lucasw/dynamic_reconfigure_tools/blob/master/dynamic_reconfigure_tools/examples/example_server.rs) and it mostly works (with lots of boiler plate to move somewhere else later), except I have to publish parameter_description and parameter_updates in a loop instead of latched as they ought to be, and the update interrupts and overwrites whatever the user was doing in the rqt dynamic reconfigure client (though even latched I need to figure out the right time to send the parameter update, maybe right in the service callback or immediately after).

Carter12s, 2024-08-02:
Yup great callout, we defo should support that.

I'd like to support it for both the ros1 and rosbridge backends with the same API, but should be easy to add.

Probably won't be able to tackle for a few days, but will get to this soon.

Carter12s, 2024-08-02:
I wonder if we should use a builder pattern for the API or if that would be more confusing....

We could do something like:

```rust
nh.advertise().latch().await;
```

Do we care about being able to turn latching on/off after a publisher is created? - Probably not...

Maybe it should be .advertise_latched()??

Carter12s, 2024-08-03:
MR for this fix is up I'll merge after CI passes.

After playing around with the API and understanding the implications of calling .advertise() multiple times (were copying ROS1's cpp API but I don't love it), I decided to add latching as another required parameter to advertise(), this will be a breaking API change, but I think a pretty reasonable one at this stage.

PR #183 (2024-08-03): Add latching behavior to native publishers
## Description
- Plumbed through the latching behavior for ROS1 native publishers
- Cleaned up the internal code structure of Publication
- Fixed a few warnings we'd been ignoring for a while
- Added integration tests to confirm latching behavior for publishers is good

## Fixes
Closes: #182 

## Checklist
- [x] Update CHANGELOG.md
closed 2024-08-03 by Carter12s