RosLibRust/roslibrust #222: Message resolution fails

wjzz, 2025-01-28:
Hello guys! I'm currently trying to introduce Rust in our C++/Python-based ROS1 codebase and so far using `roslibrust` has been great.

Today, however, my code started to fail during message resolution. 

I was able to reproduce a similar error by cloning main:

```
git clone git@github.com:RosLibRust/roslibrust.git
cd roslibrust
cargo check
```

This fails with

```
error: failed to run custom build command for `example_package v0.1.0 (/home/wjzz/Work2/roslibrust/example_package)`

Caused by:
  process didn't exit successfully: `/home/wjzz/roslibrust/target/debug/build/example_package-8905da87bb6d56cd/build-script-build` (exit status: 1)
  --- stderr
  Error: SimpleError { err: "Unable to resolve dependencies after reaching search limit.\nThe following messages have unresolved dependencies: [\"test_msgs/Float64Stamped\"]\nThese messages likely depend on packages not found in the provided search paths." }
```

While the message type is different, I am experiencing the same issue in our code with roslibrust. The same message definitions work OK with C++.

wjzz, 2025-01-28:
Update: my local issue was resolved by installing missing dependencies. It turns out that the C++ build was also failing after all.

closed 2025-01-28 by wjzz

Carter12s, 2025-01-28:
@wjzz Glad you got your issue sorted.

Always looking for perspectives from new users on what is hard about getting started, and where documentation can be improved!

wjzz, 2025-01-29:
@Carter12s Thank you! So far my experience is very good. We have a ROS1-based project and we are trying to replace some nodes with Rust as an experiment. I tried some other libraries for using Rust with ROS but they were no longer maintained and it was difficult to get started. I'm very grateful for this project and I love that you support different protocols as we are considering switching to Zenoh in the near future for at least some nodes.

So far the only gotcha I had was when I forgot to include the macro for generating messages at the top of the file - just my oversight. Up to now everything works great! Thanks for the great job!